riscv-non-isa/riscv-iommu #602: Incorrect header when a table crosses pages

vacantron, 2025-07-31:
Hi,

I found an issue where AsciiDoc automatically repeats the header row when a table crosses pages, but it seems to be repeating the incorrect header (the `fctl.GXL` in the next page should be 1).

<p align="center">
[image]
</p>

I've made some modifications based on [the privileged specification](https://github.com/riscv/riscv-isa-manual/blob/664def3c2070f7161572303a287ffa4f7abb017c/src/machine.adoc?plain=1#L1751) to fix this. Please check: https://github.com/vacantron/riscv-iommu/commit/5c9ea4f18c39a3eadd8a8779abccae939ab0d996

These are the updated previews:

<p align="center">
[image]
</p>

<p align="center">
[image]
</p>

ved-rivos, 2025-07-31:
@vacantron - Thanks. Is there a way to do it without making the source code unreadable?

vacantron, 2025-07-31:
I found the last line of each subgroup has a little bit off at the first column. I'm not sure if it's the font issue or other problem.

ved-rivos, 2025-07-31:
@vacantron - thanks. I think its best to just split the tables into two tables. I pushed a commit [here](https://github.com/riscv-non-isa/riscv-iommu/pull/569/commits/ab535df22a15f86d92dd4fd4fa3d0fa3b1466ff7) to do that.