riscv-non-isa/riscv-iommu #603: Shadow Stack page (SSP) protections for IO transactions

HaoyuanLiuHZPH, 2025-08-01:
The sentence on page 35 states 

[image]

[image]

: I would like to raise some points regarding this: 
1. Currently, the ssp encoding will only raise fault for second stage translation. But from a security standpoint, any IO requests should be explicitly forbided  from writing Shadow-Stack pages regardless of privilege level or translation stage (first or second).  

[image]

[image]
And if IO requests, like those from DMA,  is classified as implicit accesses(which I believe is), then read is also not allowed.

 2. “IO device access to Shadow Stack fault.”should be assigned a cause code other than "reserved coding 010" as in Table 11 (page 54), and should be specifically mentioned. If my understanding is correct, it should be somthing like:
"if any read transaction trying to access shadow stack page, it should raise cause 5 (read access fault) . if any write transaction trying to access shadow stack page, it should raise cause 7 (Write/AMO access fault).  Misaligned accesses to shadow stack  should raise  access-fault exception instead of address-misaligned exception   "

[image]

[image]

[image]

cyyself, 2025-08-01:
From my viewpoint, shadow stack pages should not be mapped to I/O devices for security purposes. Thus, pte.xwr = 010 should not be assigned to Shadow Stack Page in IOMMU, as mentioned in RISC-V CFI:

> The pte.xwr=010 encoding, as specified by the Zicfiss [6] extension for the Shadow Stack page type in single-stage and VS-stage page tables, remains a reserved encoding for IO transactions.

However, in IOMMU spec we have:

> To perform address translation and memory protection the IOMMU uses same page table formats as used by the CPU’s MMU for the first-stage and second-stage address translation.

which mentions "same page table formats as used by the CPU's MMU". But if we defined the pte.xwr=010 remains reserved for IO transactions, the statement "IOMMU uses same page table formats as used by the CPU’s MMU" becomes contradictory.

I think write-only permissions are useful for security purposes. Intel and AMD's IOMMU implements ZLR (zero-length-read) for write-only permissions and implemented in the Linux kernel for `DMA_FROM_DEVICE` buffers. So should we remove the statement "IOMMU uses the same page table formats as used by the CPU’s MMU"?

skye1577, 2025-08-01:
In real-world use cases (e.g., network cards), we often need to grant DMA devices write-only access to certain pages. The current RISC-V IOMMU specification mandates using the same page table format as the CPU MMU, which inherently requires mapping pages with both read and write permissions. Wouldn't it make more sense to assign perm 010 in this case? Furthermore, I believe the operating system should maintain a separate table for the IOMMU from the CPU MMU so that shadow-stack pages are not mapped to DMA devices.

[image]

ved-rivos, 2025-08-01:
@HaoyuanLiuHZPH - the IOMMU specification states that the shadow stack encoding in HS/VS or G-stage page tables remain reserved for the IOMMU. A reserved PTE encoding causes a page fault of the original access type. This forbids DMA to a page mapped as shadow stack in the S or VS stage i.e. when using shared virtual addressing. When not using shared virtual addressing, there is no reason to map shadow stacks into IOMMU page tables.

@skye1577 - there is no contradiction there. Even for the CPU MMU the encoding remains when a)  menvcfg.SSE=0 remains reserved, b) V=1 and henvcfg.SSE=0, this encoding remains reserved at VS and VU. Using same page table format does not require using same page tables - so a operating system can certainly maintain a separate page table.

ved-rivos, 2025-08-03:
> from my view, the IOMMU specification should clearly distinguish the different meanings of PTE attributes on the CPU side and the IO side. 

Repeating specifications would be would have been necessary if there was a difference between meaning of PTE attributes between CPU side and IO side. With the exception of the encoding xwr=`010, the encodings and the rules are identical. For instance, the encoding xwr=110 is reserved, the PBMT encoding of `11 is reserved, the Svnapot encodings that are reserved for CPU are also reserved for IO, PTE bits 60:54 are also reserved. If there is an exception then that is explained.

The meaning of reserved and the behavior for reserved also does not need to be repeated. The section "Virtual Address Translation Process" of the privileged manual states "If pte.v=0, or if pte.r=0 and pte.w=1, or if any bits or encodings that are reserved for future standard use are set within pte, stop and raise a page-fault exception corresponding to the original access type."

There is no concept of enabling Zicfiss for IOMMU. There are no shadow stack load/store transactions. If a pte has xwr bits set to 010 (or any other reserved encoding), then an attempt to access addresses mapped through such PTEs causes a page fault or a guest page fault exception corresponding to the original access type.

HaoyuanLiuHZPH, 2025-08-05:
Regardless of SSP, doesn't the current RISC-V IOMMU lack a Write-only permission? like Intel and AMD's ZLR write only?

skye1577, 2025-08-06:
@ved-rivos if PTE.rwx = 010 is a reserve attribute in the I/O side. From a security perspective and in the real-world applications, extending write-only permission  in the I/O-side would be make sense and reasonable.

ved-rivos, 2025-08-06:
These can be discussed for future extensions. The encoding rwx=010 is a reserved for RISC-V IOMMU.