riscv-non-isa/riscv-iommu #607: High-Performance Integrated Security Solution with IOPMP

Yuan-HT, 2025-08-14:
For the IO security solution, in scenarios where both IOMMU and IOPMP are required, the commonly adopted approach is the loosely coupled solution with IOMMU cascading IOPMP. However, this implementation is inefficient and PPA-unfriendly. We may consider a tightly coupled solution by embedding IOPMP into IOMMU, retaining only the complete checker capability. Before the IOMMU performs the table lookup to obtain the PA and update the TLB, it queries the IOPMP and then writes the permission results into the TLB. This solution utilizes the most lightweight security check module, combined with the comprehensive caching capability of IOMMU, to achieve a high-performance IOMMU security solution.

This approach introduces several issues:

1. The configuration of IOMMU and IOPMP driver software, as well as page table entries.

2. The refresh mechanism of IOMMU when IOPMP entries are updated.（Referring to the MFENCE capability of SMMT in the MMU, should we incorporate a FENCE-based FLUSH function? Currently, IOMMU flushing is implemented via IOFENCE.C in the CQ）

3. The error reporting mechanism, where the two share INT and MIS.

ved-rivos, 2025-08-15:
I hope that addressed the question . To summarize the PMA/PMP functionality is architecturally part of the iobridge and not the iommu. The programming interface for this is not specified by the iommu. Standardizing the programming interface for PMA and/or PMP functions maybe part of a future effort. This does not restrict clever implementations that fuse the capabilities . Please ask if there are further questions .

closed 2025-08-15 by ved-rivos

Yuan-HT, 2025-08-15:
Perhaps my description was not clear enough. What I want to discuss is the combined solution of IOMMU and IOPMP, rather than the PMP/PMA described in the IOMMU specification. It is obvious that the IOPMP specification has stronger I/O isolation capabilities, and PMP/PMA is not the main focus of my discussion. Below, I will elaborate on what I consider to be an efficient solution.

# First, let's discuss why I'm mentioning IOPMP here instead of PMP
 If we refer to the on-core PMP, which consists of 16 registers and can return results within a few cycles per query, it's clear that this lightweight design causes almost no performance loss or excessive area overhead. However, the security protection capability of the PMP scheme is only moderate, so there's no need to discuss it here. What I hope to explore is a collaborative scheme between IOPMP and IOMMU in the context of complex large SoCs. These two components are specified in different standard manuals, and the current standards offer little support for the tightly coupled scheme—the combined approach I initially described. I aim to clarify this, as it would be beneficial for improving the PPA (Power, Performance, Area) of a complex SoC architecture that includes both IOMMU and IOPMP.

# Then discuss the rationality of the tight coupling scheme
Assume that there are 16 PCIe channels in our SoC. If we want to implement an IOMMU for virtualization management, a relatively good solution is to use 1 PTW (Page Table Walker) for page table virtualization translation, along with 4/8/16 TLBs and corresponding IO bridges. Each group of TLB + IO bridge manages 4/8/16 PCIe devices respectively. When there are a large number of devices, their throughput is high and they are distributed in different locations of the SoC. Managing them separately can improve throughput; sharing a PTW can reduce the area, and the cache size of a single TLB can also be slightly reduced.

At this point, if we consider adding IOPMP, and if the devices are directly connected to a NoC (Network on Chip) through the IO bridge, our IOPMP must be split into 4/8/16 units, with each IO bridge equipped with an IOPMP for inspection. Although each IOPMP can adopt a lightweight scheme (such as the rapid-k mode), its resource usage and impact on bus performance still exist. In industrial scenarios, IOMMU solutions with relatively small performance loss result in a throughput loss of around 5%. On top of this 5% loss, how much additional throughput loss will be caused by the new inspections introduced by IOPMP?

Moreover, in this scenario, an optimization scheme for IOPMP is to separate the checkers that query the srcmd, mdcfg, and entry page tables into a shared module, while distributing the IO bridge processing across the SoC. In this microarchitecture, the IOMMU and IOPMP are closely related. Imagine integrating the PTW of the IOMMU with the checker of the IOPMP, and combining their TLBs and IO bridges. This can not only reduce the area but also optimize the impact on transmission performance. The typical latency for the IOMMU to perform a first-level table lookup and fetch data from DDR is 200 cycles. For IOPMP in the most complex mode (format-0) with minimal area (using SRAM for storage and fully serial entry queries), the typical latency is 30 cycles. The total latency for 3-level page table translation by the IOMMU plus permission query by the IOPMP is 630 cycles. At this point, the IOMMU queries the PA (Physical Address) and then checks the PA permission in the IOPMP, which is then stored in the TLB. New permission attributes and corresponding refresh mechanisms are added to the TLB, enabling the fastest PA-VA-security checks in each IO bridge path. I believe this approach is reasonable. (Of course, there are still other combinations, such as using only IOMMU or only IOPMP.)

# Finally, discuss the details of the current spec conflicts
1. When a permission error occurs, the specifications of IOMMU and IOPMP handle IOBRIDGE differently. Although these are just some implementation details, they violate the spec.
2. How to handle the refresh request generated after the IOPMP update requires the addition of an IOMMU refresh mechanism.
> I am not sure what INT and MIS are.
3. For error reporting, when IOPMP has only a checker, it does not support MSI interrupts and only retains line interrupts, which need to be merged with the interrupts of IOMMU. @ved-rivos