riscv-non-isa/riscv-iommu #630: read address translated in mrif mode

1252005989, 2025-09-16:
Hi, there is a further question about issue 625.
https://github.com/riscv-non-isa/riscv-iommu/issues/625
What is the expcted process flow when a read request address is transalated in mrif mode? Still read mrif address as normal?

ved-rivos, 2025-09-16:
In MRIF mode, the read and writes are handled either by the IOMMU or by the I/O Bridge. The details are specified by the [AIA specification](https://github.com/riscv/riscv-aia). 

- Accesses that are not four byte wide and naturally aligned to 4-byte boundary are discarded.
- A naturally aligned 32-bit read, is returned a value of 0. 
- A naturally aligned 32-write that is not interpreted as a MSI is discarded.
- A naturally aligned 32-write that is a MSI updates the MRIF.

ved-rivos, 2025-09-16:
It’s preferable to return 0. Aborting is acceptable but not preferred

123wh123, 2025-10-01:
thanks sir