riscv-non-isa/riscv-iommu #639: Question about requirement of changing device directory table entry

viktoryou, 2025-09-29:
From the spec,
> If software changes a leaf-level DDT entry (i.e, a device context (DC), of device with device_id = D)
then the following invalidations must be performed:
> * `IODIR.INVAL_DDT` with `DV=1` and `DID=D`
> * If `DC.iohgatp.MODE != Bare`
>   * `IOTINVAL.VMA` with `GV=1`, `AV=PSCV=0`, and `GSCID=DC.iohgatp.GSCID`
>   * `IOTINVAL.GVMA` with `GV=1`, `AV=0`, and `GSCID=DC.iohgatp.GSCID`
> * else
>   * If `DC.tc.PDTV==1`
>     * `IOTINVAL.VMA` with `GV=AV=PSCV=0`
>   * else if `DC.fsc.MODE != Bare`
>     * `IOTINVAL.VMA` with `GV=AV=0` and `PSCV=1`, and `PSCID=DC.ta.PSCID`

I think the sequence should be more precise as
> If software changes a leaf-level DDT entry (i.e, a device context (DC), of device with device_id = D)
then the following invalidations must be performed:
> * `IODIR.INVAL_DDT` with `DV=1` and `DID=D`
> * If `DC.iohgatp.MODE != Bare`
>   * if `DC.tc.PDTV==1`
>     * `IOTINVAL.VMA` with `GV=1`, `AV=PSCV=0`, and `GSCID=DC.iohgatp.GSCID`
>     * `IOTINVAL.GVMA` with `GV=1`, `AV=0`, and `GSCID=DC.iohgatp.GSCID`
>   * else if `DC.fsc.MODE != Bare`
>     * `IOTINVAL.VMA` with `GV=1`, `AV=0`, `PSCV=1`, `GSCID=DC.iohgatp.GSCID` and `PSCID=DC.ta.PSCID`
>     * `IOTINVAL.GVMA` with `GV=1`, `AV=0`, and `GSCID=DC.iohgatp.GSCID`  
>   * else
>     * `IOTINVAL.GVMA` with `GV=1`, `AV=0`, and `GSCID=DC.iohgatp.GSCID` 
> * else
>   * If `DC.tc.PDTV==1`
>     * `IOTINVAL.VMA` with `GV=AV=PSCV=0`
>   * else if `DC.fsc.MODE != Bare`
>     * `IOTINVAL.VMA` with `GV=AV=0` and `PSCV=1`, and `PSCID=DC.ta.PSCID`

This improvement seems compatible with the spec, and provides the least requirement to the software. Please correct me if I had missed anything.

ved-rivos, 2025-09-29:
The guidance presently will encompass all of these cases. For instance, when PDTV is 0, the only PSCID is DC.ta.PSCID and hence the guideline to issue "IOTINVAL.VMA with GV=1, AV=PSCV=0, and GSCID=DC.iohgatp.GSCID" is equivalent and does not lead to any additional invalidations than necessary. Likewise when PDTV=0 and DC.fsc.MODE == Bare, the "IOTINVAL.VMA with GV=1, AV=PSCV=0, and GSCID=DC.iohgatp.GSCID" is effectively a no-op since there are no first stage PTEs cached. Doing the fine grained analysis as proposed leads to additional SW but may not lead any significant benefits.

ved-rivos, 2025-09-30:
For the case where iohgatp.mode is Bare you suggested:

```
1. if DC.tc.PDTV==1
     a. IOTINVAL.VMA with GV=1, AV=PSCV=0, and GSCID=DC.iohgatp.GSCID
     b. IOTINVAL.GVMA with GV=1, AV=0, and GSCID=DC.iohgatp.GSCID
2. else if DC.fsc.MODE != Bare
    c. IOTINVAL.VMA with GV=1, AV=0, PSCV=1, GSCID=DC.iohgatp.GSCID and PSCID=DC.ta.PSCID
    d. IOTINVAL.GVMA with GV=1, AV=0, and GSCID=DC.iohgatp.GSCID
3. else
   e. IOTINVAL.GVMA with GV=1, AV=0, and GSCID=DC.iohgatp.GSCID
```

The case 1a and 1b are the same as the existing guidance.

For case 2, the difference between 1a and 2c is not meaningful since when PDTV is 0 and DC.fsc.mode !=Bare, there is only one address space identified by DC.ta.PSCID. 2c tries to identify that precisely but 1a would do that same i.e. 1a and 2c are equivalent for this case.

For case 3, you suggested that IOTINVAL.VMA can be avoided. That is true. However, when PDTV is 0 and DC.fsc.Mode is Bare, there is actually nothing cached for VS-stage. So the IOTINVAL.VMA would effectively be a nop. 

Hence the existing guidance does not do any extra invalidation than necessary. However, it does simply SW in not having to analyze and do different actions based on the 3 cases.

viktoryou, 2025-10-01:
Got that point. Thanks for the detailed explanation.